yuq/mesa-lima #13: Implement glScissor support

PabloPL, 2017-12-18:
It's mostly done on https://github.com/yuq/mesa-lima/pull/10.
There is only one case to handle - when scissors = (0,0,0,0)

Limadriver in this case is not doing any scissors and displaying full/normal image (probably a bug)
Mali binary driver is generating only following plbu commands (dump from egl_quad_flat from limadrver tests)
>>0x00000200 0x1000010b // PRIMITIVE_SETUP
0x00010000 0x1000010c // BLOCK_STEP
0x1d001d00 0x10000109 // TILED_DIMENSIONS
0x0000001e 0x30000000 // BLOCK_STRIDE
0x400f8400 0x280001c1 // ARRAY_ADDRESS
0x00000000 0x10000107 // VIEWPORT_X
0x43f00000 0x10000108 // VIEWPORT_W
0x00000000 0x10000105 // VIEWPORT_Y
0x43f00000 0x10000106 // VIEWPORT_H
0x40040000 0x10000103 // TILE_HEAP_START
0x40070000 0x10000104 // TILE_HEAP_END
0x00000000 0x50000000 // END

PabloPL, 2017-12-19:
And this is what is generated by mesa currently
```
lima: dump command stream enabled
num config 1
gbm format 34325241
lima: update attribute info at va 43800
0000: 0x0008c000 0x00012002 0x0008c00c 0x00012003
lima: update varying info at va 43000
0000: 0x00041000 0x00008020 0x00040000 0x00008003
lima add vs cmd at va 45000
0000: 0x00028000 0x50000000 0x00000001 0x50000000
0010: 0x00044000 0x30000000 0x00042000 0x40020000
0020: 0x00100400 0x10000040 0x01000100 0x10000042
0030: 0x00000003 0x10000041 0x00043800 0x20040000
0040: 0x00043000 0x20040008 0x04000000 0x00000000
0050: 0x00000000 0x60000000 0x00000000 0x50000000
lima add plbu cmd at va 45800
0000: 0x00000200 0x1000010b 0x00000000 0x1000010c
0010: 0x0f000f00 0x10000109 0x00000010 0x30000000
0020: 0x00042800 0x280000ff 0x00000000 0x10000107
0030: 0x43800000 0x10000108 0x00000000 0x10000105
0040: 0x43800000 0x10000106 0x00010002 0x60000000
0050: 0x00002200 0x1000010b 0x000498c0 0x80004100
0060: 0x00000000 0x1000010a 0x00000000 0x1000010e
0070: 0x3f800000 0x1000010f 0x04000000 0x00060000
0080: 0x00010001 0x60000000 0x00000000 0x00000000
lima: add render state at va 498c0
0000: 0x00000000 0x00000000 0xfc3b1ad2 0x0000003e
0010: 0xffff0000 0xff000007 0xff000007 0x00000000
0020: 0x0000f807 0x00048003 0x00000000 0x00000000
0030: 0x00000000 0x00000302 0x00003000 0x00040000
0
lima: update attribute info at va 43840
0000: 0x0008c090 0x00006002 0x0008c0b4 0x00008003
lima: update varying info at va 43040
0000: 0x00041040 0x00008020 0x00040040 0x00008003
lima add vs cmd at va 45060
0000: 0x00028000 0x50000000 0x00000001 0x50000000
0010: 0x00044000 0x30030000 0x00042040 0x40050000
0020: 0x00301000 0x10000040 0x01000100 0x10000042
0030: 0x00000003 0x10000041 0x00043840 0x20040000
0040: 0x00043040 0x20040008 0x03000000 0x00000000
0050: 0x00000000 0x60000000 0x00000000 0x50000000
lima add plbu cmd at va 45888
0000: 0x00010002 0x60000000 0x00002200 0x1000010b
0010: 0x00049900 0x80004104 0x007f8100 0xffffe000
0020: 0x00000000 0x1000010a 0x00000000 0x1000010e
0030: 0x3f800000 0x1000010f 0x03000000 0x00040000
0040: 0x00010001 0x60000000 0x00000000 0x00000000
lima: add render state at va 49900
0000: 0x00000000 0x00000000 0xfc3b1ad2 0x0000003e
0010: 0xffff0000 0xff000007 0xff000007 0x00000000
0020: 0x0000f807 0x00048043 0x00000000 0x00000000
0030: 0x00000000 0x00000302 0x00003000 0x00040040
gp submit wait error
lima varying dump at va 40000
0000: 0.000000 0.000000 0.000000 0.000000
0010: 0.000000 0.000000 0.000000 0.000000
0020: 0.000000 0.000000 0.000000 0.000000
0030: 0.000000 0.000000 0.000000 0.000000
0040: 0.000000 0.000000 0.000000 0.000000
0050: 0.000000 0.000000 0.000000 0.000000
0060: 0.000000 0.000000 0.000000 0.000000
0070: 0.000000 0.000000 0.000000 0.000000
0080: 0.000000 0.000000 0.000000 0.000000
0090: 0.000000 0.000000 0.000000 0.000000
00a0: 0.000000 0.000000 0.000000 0.000000
00b0: 0.000000 0.000000 0.000000 0.000000
00c0: 0.000000 0.000000 0.000000 0.000000
00d0: 0.000000 0.000000 0.000000 0.000000
00e0: 0.000000 0.000000 0.000000 0.000000
00f0: 0.000000 0.000000 0.000000 0.000000
lima gl_pos dump at va 41000
0000: -1.000000 -1.000000 1.000000 1.000000
0010: -1.000000 -1.000000 1.000000 1.000000
0020: -1.000000 -1.000000 1.000000 1.000000
0030: -1.000000 -1.000000 1.000000 1.000000
0040: 0.000000 0.000000 0.000000 0.000000
0050: 0.000000 0.000000 0.000000 0.000000
0060: 0.000000 0.000000 0.000000 0.000000
0070: 0.000000 0.000000 0.000000 0.000000
0080: 0.000000 0.000000 0.000000 0.000000
0090: 0.000000 0.000000 0.000000 0.000000
00a0: 0.000000 0.000000 0.000000 0.000000
00b0: 0.000000 0.000000 0.000000 0.000000
00c0: 0.000000 0.000000 0.000000 0.000000
00d0: 0.000000 0.000000 0.000000 0.000000
00e0: 0.000000 0.000000 0.000000 0.000000
00f0: 0.000000 0.000000 0.000000 0.000000
gbm-surface: main.c:289: Render: Assertion `glGetError() == GL_NO_ERROR' failed.
Aborted
```
@yuq Any ideas

yuq, 2017-12-19:
PLBU cmd is necessary for every frame. You may try to generate binary mali driver like PLBU cmd when scissor is 0.

PabloPL, 2017-12-19:
I've already done this and posted in first comment (with some comments).

yuq, 2017-12-19:
BTW. do we need to do anything if scissor is 0?

PabloPL, 2017-12-19:
I think we should render empty screen (since we set that we don't want anything to display).

yuq, 2017-12-19:
To my understanding, if set scissor to 0, no following draw should apply to the frame buffer, including the glClear(). So this frame buffer should remain the same as last none-zero draw result. Can you make an experiment by running the scissor test on a normal desktop with AMD/NV mesa driver to see what's expected result? If we need do nothing, we may skip the draws when scissor is 0.

PabloPL, 2017-12-19:
Made a test using mali binary driver - it rendered an empty screen (dumped using mali syscall tracker).
Will do the same with intel in a few minutes (since this is gpu which i have on my laptop) and update comment.

PabloPL, 2017-12-19:
I'm not happy with current implementation, but maybe we could merge it and leave a comment on wiki that this edge case is not supported yet (since there are probably more important things to implement).

yuq, 2017-12-19:
Then we can just skip draws when scissor=0 for now.

PabloPL, 2017-12-19:
> Then we can just skip draws when scissor=0 for now.
Implemented. Now it's working for me, for all cases.

PabloPL, 2017-12-29:
Working implementation (for both zero and non zero scissor) in https://github.com/yuq/mesa-lima/pull/15.

> Then we can just skip draws when scissor=0 for now.

It turns out that we can't skip drawing.
We need to
- create plb cmd header (if not already created) and skipp all other plb cmds
- skip creating vs cmd list
Then it's working fine.

PabloPL, 2017-12-31:
TODO
 - we should set scissor (non zero case) only for the first draw that enabled the scissor and for latter draw only if scissor is dirty
- check why scissor is not affecting bounds of region cleared by glClear