yuq/mesa-lima #16: Implement glBlend

PabloPL, 2018-01-02:


PabloPL, 2018-01-03:
This functionality looks working from begining (when those two functions were added), it was missing only special handling for blending disabled case.
I've tested few images generated by mesa and compared them with the same code on intel gpu.

I've also compared values generated by lima_blend_func and lima_blend_factor with value generated by binary shader and they look exactly the same.

@yuq Did You found any cases where it wasn't working (and that's why it was disabled) ?
Otherwise, PR for this task is in https://github.com/yuq/mesa-lima/pull/17

PabloPL, 2018-01-03:
TODO:
Looking at lima-ng sources alpha_blend could containt (at first byte) color mask.
Need to check it (and fix either now or in next patch).

yuq, 2018-01-04:
Yes, it doesn't work properly when blend disabled and there's something need to be investigated (comment in the code), so I disable it.

PabloPL, 2018-01-05:
Confirmed - color mask value is also stored in alpha_blend field.
Created separated PR for this https://github.com/yuq/mesa-lima/pull/18.

Is there anything else done with this issue or we can close it (i wasn't able to find any information about independ blend support :/).

yuq, 2018-01-06:
I think no more for blend now. We can handle only zbuf case after we have depth & stencil support. Thanks for your work.

closed 2018-01-06 by PabloPL