yuq/mesa-lima #20: Banana PI R2 / Mediatek MT7623 support ?

jfdelnero, 2018-01-10:
Hi,

I am trying to make mesa-lima working on the Banana PI R2. This board is based on the MT7623 and have a Mali 450 GPU.

The problem is that it seems that the Mediatek DRM driver included in the board's kernel doesn't seem to expose the render node (no /dev/dri/renderD128 available).
I will probably have to patch/change the kernel's DRM configuration.

What is the right DRM driver to use with mesa-lima ?

Best Regards,
Jean-François DEL NERO

yuq, 2018-01-11:
I guess you're using the DRM driver for the display module of MT7623. Mali GPU has its own DRM driver at: https://github.com/yuq/linux-lima

For off-screen rendering, you just need the Mali GPU DRM driver, while for on-screen rendering you need both Mali GPU DRM and display DRM for MT7623. For on screen rendering mesa-lima only support sun4i display DRM driver now.

jfdelnero, 2018-01-11:
Thanks !
This appears to be the full kernel. Is there a way to get just the driver or the patches ?

yuq, 2018-01-11:
I just pick up the 4.13 kernel then apply my patches on top of it. So start from the 4.13 release commit, all commits after are the driver (except some patches not mention drm/lima). But you may have to do some port work if your kernel is not 4.13.

jfdelnero, 2018-01-11:
yes, the BPI R2/mediatek kernel is the 4.4.70.
So i will have to generate the patches between the 4.13 and the lima 4.13 and try to apply them to the mediatek kernel.

anarsoul, 2018-01-13:
@jfdelnero I think you need to port mediatek patches for 4.4.70 to 4.13. It could be pretty hard to backport lima patches from 4.13 to 4.4 kernel. DRM subsystem has changed a lot from 4.4 to 4.13.

jfdelnero, 2018-01-13:
Yes i know. That's why i am currently trying this : 
https://github.com/frank-w/BPI-R2-4.14