yuq/mesa-lima #32: Blue/Red channels of texture are swapped

anarsoul, 2018-02-26:
With the FBO test app, seems texture support has some problem
Blue/Red channel swap: test app render buffer is PIPE_FORMAT_B8G8R8A8_UNORM, while texture is PIPE_FORMAT_R8G8B8A8_UNORM, the final result is B/R channel swap

anarsoul, 2018-03-14:
@yuq I don't understand how lima chooses render buffer formats, looks like it always uses RGBA8888 regardless of what app is specified, see: https://github.com/yuq/mesa-lima/blob/lima-17.3/src/gallium/drivers/lima/lima_draw.c#L1003

So far it doesn't look like a bug in texturing code, but I can fix it.

yuq, 2018-03-14:
Oh right, so this get the first draw of FBO test wrong format. You may get the render buffer format from ctx->framebuffer->cbuf.

PR #35 (2018-03-16): lima: properly support RGBA render targets
Render target is BGRA by default, so we need to set swap
channel bit in mrt_bits for RGBA.

Fixes #32
closed 2018-03-16 by yuq