yuq/mesa-lima #36: Increase gp tile heap size

anarsoul, 2018-03-15:
'glmark2-drm -b build' runs out of tile heap:

[  318.835774] lima 1c40000.gpu: gp error irq state=4 status=2b

where state=4 is LIMA_GP_IRQ_PLBU_OUT_OF_MEM. limare has 1mb tile heap size while we are at 32kb.

Increasing heap size to 1mb fixes this particular issue, but now it gets mmu page fault while still rendering black screen:

[ 4212.840241] lima 1c40000.gpu: mmu page fault at 0x1b002740 from bus id 0 of type read on ppmmu0
[ 4212.857258] lima 1c40000.gpu: mmu page fault at 0xf002780 from bus id 0 of type read on ppmmu1
[ 4212.875500] lima 1c40000.gpu: mmu resume
[ 4212.887884] lima 1c40000.gpu: mmu resume

anarsoul, 2018-03-22:
After 637f26403627f806f10d92ece71d43215b161400 it almost works, sometimes horse's just stuck (no rotating) and I get following in dmesg:

[  195.830248] lima 1c40000.gpu: pp error irq state=201 status=40
[  195.845414] lima 1c40000.gpu: pp error irq state=201 status=40,

state=201 is LIMA_PP_IRQ_INVALID_PLIST_COMMAND | LIMA_PP_IRQ_END_OF_FRAME

yuq, 2018-03-23:
Nice to hear it, it's a little surprise that glmark2 build can draw as depth and index draw are not supported yet, maybe it doesn't use them.

As the stuck, it's like the problem I'm looking at for mali450 https://github.com/yuq/mesa-lima/issues/26, I think may be a cache problem because I see sometime this pp irq error and sometimes MMU fault.

yuq, 2018-03-23:
@anarsoul did you modified glmark2? I try to run with 'glmark2-drm -b build' but get:
Error: eglChooseConfig() didn't return any configs
Error: Error: Couldn't get GL visual config!
Error: main: Could not initialize canvas

BTW. my glamrk2 is from the github.

anarsoul, 2018-03-23:
I did. I commented out "EGL_DEPTH_SIZE, requested_visual_config_.depth" in glmark2/src/gl-state-egl.cpp

anarsoul, 2018-03-23:
https://photos.app.goo.gl/q0Q4RRjeHFWceFdf2

yuq, 2018-03-23:
Great!

After comment out the depth, glmark2 can continue but fail at:
Error: Glmark2 needs OpenGL(ES) version >= 2.0 to run (but version string is: '1.4 Mesa 17.3.0 (git-637f264036)')!
Error: main: Could not initialize canvas

So you should also did some change on mesa.

yuq, 2018-03-23:
I can run with glmark2-es2-drm. So lima mesa export OpenGL 1.4 and OpenGL ES 2.0.

@enunes I think that's why the piglit not run OpenGL 2 tests, we need to add some missing feature/ext for mesa exporting OpenGL 2.0 support (I remember I did for OpenGL ES 2.0).

anarsoul, 2018-03-23:
@yuq I think we need occlusion query (at least a stub) to advertise GL 2.0

anarsoul, 2018-03-23:
Fixed by 470b1610a65af011dcba0c1ea1c652885cf06c26

closed 2018-03-23 by anarsoul

superna9999, 2018-04-05:
Hi,

I integrated mesa-lima 18.0, and linux-lima 4.16-rc into meta-meson, kmscube and glmark2 runs (with some tweaks):
https://github.com/superna9999/meta-meson/tree/topic/mesa-lima

it can be useful to test on Amlogic SoCs !

yuq, 2018-04-05:
Thanks @superna9999, I've added your metadata info into the wiki.