yuq/mesa-lima #38: sometimes ir compiler will deadloop (discovered by piglit)

Icenowy, 2018-03-28:
mesa-lima commit: 2a54f24d4452475a605909e16287b5c2f057758c
piglit commit: af1140c9d152e11380519ff79dde2cac110bb327
piglit test: shaders/glsl-deadcode-varying

The test hangs until the process is manually terminated.

stacktrace gained by sending SIGABRT to the shader_runner:
```
#0  gpir_get_min_end (node=node@entry=0x12af830) at ir/gp/scheduler.c:300
#1  0xb63fbcfc in schedule_try_place_node (instr=instr@entry=0x12a9f48, node=node@entry=0x12af830) at ir/gp/scheduler.c:354
#2  0xb63fc7a8 in gpir_sched_instr_pass (ready_list=0xbefcd1f4, instr=<optimized out>) at ir/gp/scheduler.c:444
#3  schedule_one_instr (ready_list=0xbefcd1f4, block=0x1274958) at ir/gp/scheduler.c:564
#4  schedule_block (block=0x1274958) at ir/gp/scheduler.c:604
#5  gpir_schedule_prog (comp=0xb64c7368, comp@entry=0x12aa8c8) at ir/gp/scheduler.c:797
#6  0xb63fa2f0 in gpir_compile_nir (prog=prog@entry=0x1277550, nir=nir@entry=0x1276f78) at ir/gp/nir.c:389
#7  0xb63f7f54 in lima_create_vs_state (pctx=<optimized out>, cso=<optimized out>) at lima_program.c:279
#8  0xb61f4e38 in st_create_vp_variant (key=0xbefcd2c8, stvp=0x12a2510, st=0x126f2b8) at state_tracker/st_program.c:567
#9  st_get_vp_variant (st=st@entry=0x126f2b8, stvp=stvp@entry=0x12a2510, key=key@entry=0xbefcd2c8) at state_tracker/st_program.c:623
#10 0xb61adf00 in st_update_vp (st=0x126f2b8) at state_tracker/st_atom_shader.c:191
#11 0xb61aa8a0 in st_validate_state (st=st@entry=0x126f2b8, pipeline=pipeline@entry=ST_PIPELINE_RENDER) at state_tracker/st_atom.c:251
#12 0xb61c8e40 in prepare_draw (ctx=0x1279030, st=0x126f2b8) at state_tracker/st_draw.c:122
#13 st_draw_vbo (ctx=ctx@entry=0x1279030, prims=0xbefcd3b0, prims@entry=0xbefcd3a8, nr_prims=nr_prims@entry=1, ib=0x0, 
    index_bounds_valid=index_bounds_valid@entry=1 '\001', min_index=<optimized out>, max_index=<optimized out>, max_index@entry=3, tfb_vertcount=0x0, stream=0, 
    indirect=0x0) at state_tracker/st_draw.c:148
#14 0xb61903c4 in vbo_draw_arrays (ctx=ctx@entry=0x1279030, mode=mode@entry=5, start=start@entry=0, count=count@entry=4, numInstances=numInstances@entry=1, 
    baseInstance=baseInstance@entry=0, drawID=drawID@entry=0) at vbo/vbo_exec_array.c:486
#15 0xb61906d8 in vbo_exec_DrawArrays (mode=5, start=0, count=4) at vbo/vbo_exec_array.c:641
#16 0xb6d8f268 in stub_glDrawArrays (mode=5, first=0, count=4) at /root/piglit/build/tests/util/piglit-dispatch-gen.c:12150
#17 0xb6e54be4 in piglit_draw_rect_from_arrays (verts=0xbefcd4a0, tex=0x0, use_patches=false, instance_count=1) at /root/piglit/tests/util/piglit-util-gl.c:680
#18 0xb6e55154 in piglit_draw_rect_custom (x=-1, y=-1, w=2, h=2, use_patches=false, instance_count=1) at /root/piglit/tests/util/piglit-util-gl.c:801
#19 0xb6e55198 in piglit_draw_rect (x=-1, y=-1, w=2, h=2) at /root/piglit/tests/util/piglit-util-gl.c:810
#20 0x00015218 in piglit_display () at /root/piglit/tests/shaders/shader_runner.c:3046
#21 0xb6e84238 in enter_event_loop (winsys_fw=0x12658a0) at /root/piglit/tests/util/piglit-framework-gl/piglit_gbm_framework.c:42
#22 0xb6e83884 in run_test (gl_fw=0x12658a0, argc=2, argv=0xbefcdab4) at /root/piglit/tests/util/piglit-framework-gl/piglit_winsys_framework.c:88
#23 0xb6e61808 in piglit_gl_test_run (argc=2, argv=0xbefcdab4, config=0xbefcd924) at /root/piglit/tests/util/piglit-framework-gl.c:229
#24 0x0000c5a8 in main (argc=2, argv=0xbefcdab4) at /root/piglit/tests/shaders/shader_runner.c:61
```

Icenowy, 2018-03-28:
shaders/glsl-fs-convolution-1 also fails in schedule_try_place_node()

backtrace:
```
#0  gpir_instr_insert_alu_check (node=0xd30308, instr=0xd3f648) at ir/gp/instr.c:84
#1  gpir_instr_try_insert_node (instr=instr@entry=0xd3f648, node=node@entry=0xd30308) at ir/gp/instr.c:377
#2  0xb640bd10 in schedule_try_place_node (instr=instr@entry=0xd3f648, node=node@entry=0xd30308) at ir/gp/scheduler.c:355
#3  0xb640c7a8 in gpir_sched_instr_pass (ready_list=0xbef8d1e4, instr=<optimized out>) at ir/gp/scheduler.c:444
#4  schedule_one_instr (ready_list=0xbef8d1e4, block=0xca8920) at ir/gp/scheduler.c:564
#5  schedule_block (block=0xca8920) at ir/gp/scheduler.c:604
#6  gpir_schedule_prog (comp=0xb64d7368, comp@entry=0xd556f8) at ir/gp/scheduler.c:797
#7  0xb640a2f0 in gpir_compile_nir (prog=prog@entry=0xd4d890, nir=nir@entry=0xcf5c98) at ir/gp/nir.c:389
#8  0xb6407f54 in lima_create_vs_state (pctx=<optimized out>, cso=<optimized out>) at lima_program.c:279
#9  0xb6204e38 in st_create_vp_variant (key=0xbef8d2b8, stvp=0x9f83a8, st=0xa04628) at state_tracker/st_program.c:567
#10 st_get_vp_variant (st=st@entry=0xa04628, stvp=stvp@entry=0x9f83a8, key=key@entry=0xbef8d2b8) at state_tracker/st_program.c:623
#11 0xb61bdf00 in st_update_vp (st=0xa04628) at state_tracker/st_atom_shader.c:191
#12 0xb61ba8a0 in st_validate_state (st=st@entry=0xa04628, pipeline=pipeline@entry=ST_PIPELINE_RENDER) at state_tracker/st_atom.c:251
#13 0xb61d8e40 in prepare_draw (ctx=0xa0a6a0, st=0xa04628) at state_tracker/st_draw.c:122
#14 st_draw_vbo (ctx=ctx@entry=0xa0a6a0, prims=0xbef8d3a0, prims@entry=0xbef8d398, nr_prims=nr_prims@entry=1, ib=0x0, 
    index_bounds_valid=index_bounds_valid@entry=1 '\001', min_index=<optimized out>, max_index=<optimized out>, max_index@entry=3, tfb_vertcount=0x0, stream=0, 
    indirect=0x0) at state_tracker/st_draw.c:148
#15 0xb61a03c4 in vbo_draw_arrays (ctx=ctx@entry=0xa0a6a0, mode=mode@entry=5, start=start@entry=0, count=count@entry=4, numInstances=numInstances@entry=1, 
    baseInstance=baseInstance@entry=0, drawID=drawID@entry=0) at vbo/vbo_exec_array.c:486
#16 0xb61a06d8 in vbo_exec_DrawArrays (mode=5, start=0, count=4) at vbo/vbo_exec_array.c:641
#17 0xb6d9f268 in stub_glDrawArrays (mode=5, first=0, count=4) at /root/piglit/build/tests/util/piglit-dispatch-gen.c:12150
#18 0xb6e64be4 in piglit_draw_rect_from_arrays (verts=0xbef8d490, tex=0x0, use_patches=false, instance_count=1) at /root/piglit/tests/util/piglit-util-gl.c:680
#19 0xb6e65154 in piglit_draw_rect_custom (x=-1, y=-1, w=2, h=2, use_patches=false, instance_count=1) at /root/piglit/tests/util/piglit-util-gl.c:801
#20 0xb6e65198 in piglit_draw_rect (x=-1, y=-1, w=2, h=2) at /root/piglit/tests/util/piglit-util-gl.c:810
#21 0x00015218 in piglit_display () at /root/piglit/tests/shaders/shader_runner.c:3046
#22 0xb6e94238 in enter_event_loop (winsys_fw=0x9eaaa0) at /root/piglit/tests/util/piglit-framework-gl/piglit_gbm_framework.c:42
#23 0xb6e93884 in run_test (gl_fw=0x9eaaa0, argc=2, argv=0xbef8daa4) at /root/piglit/tests/util/piglit-framework-gl/piglit_winsys_framework.c:88
#24 0xb6e71808 in piglit_gl_test_run (argc=2, argv=0xbef8daa4, config=0xbef8d914) at /root/piglit/tests/util/piglit-framework-gl.c:229
#25 0x0000c5a8 in main (argc=2, argv=0xbef8daa4) at /root/piglit/tests/shaders/shader_runner.c:61
```

yuq, 2018-03-28:
Current compiler is not in a complete state, some instructions are not implemented and control flow is not supported, so I'd expect many tests of piglit should fail. We can get a list of them to implement one by one.

Icenowy, 2018-03-28:
I will report all fails and crashes in a new issue when I get all (or all except these deadloop ones) tests executed.

However, even the compiler is not complete it shouldn't deadloop.

yuq, 2018-03-28:
Yes, dead loop will block the piglit test run which causes much manual work. I see piglit has timeout function, maybe it needs some option to enable.

You can add deadloop tests to the fail test list too.

enunes, 2018-04-20:
I was not able to reproduce a deadloop with those specific tests anymore, tried shaders/glsl-fs-convolution-1 and shaders/glsl-deadcode-varying now and they just failed with:
```
shader_runner: ir/gp/scheduler.c:395: gpir_sched_node: Assertion `result' failed.
```
Does any other test still deadloop in the compiler?