yuq/mesa-lima #39: multiple fails and crashes when running piglit

Icenowy, 2018-03-28:
I have tried to run piglit with lima, and most tests (if not skipped) fail or crash.

Here's the piglit result: (running "gpu" testsuite)

[results.json.gz](https://github.com/yuq/mesa-lima/files/1856138/results.json.gz)

Note: the following tests hangs when running, I manually killed them or excluded them to continue the test:
```
glsl-fs-texture2d-dependent-4 glsl-deadcode-varying glsl-fs-convolution-1 glsl-vs-mat-sub-1 glsl-vs-mat-add-1 glsl-fs-convolution-2 glsl-vs-mat-mul-1 glsl-clamp-vertex-color glsl-vs-dot-vec2 infinity_and_nan mad add max min mul_with_swizzle_and_masking sub_with_swizzle gl-2.0-large-point-fs gl_rgba8-render-sanity
```

yuq, 2018-03-28:
Thanks, I see the results: 486/3484.

Icenowy, 2018-03-28:
There's also some programs that uses depth/stencil buffer which will fail to eglChooseConfig and be considered as skipped, and I manually ignored some hanging tests, so the real test number suitable for Mali Utgard is bigger than 3484.

Icenowy, 2018-03-29:
@yuq is there any debugging environment variables that I can use to run piglit again (to capture more details)?

yuq, 2018-03-29:
For lima you can find two:
LIMA_SHADER_DEBUG
LIMA_DUMP_COMMAND_STREAM
https://github.com/yuq/mesa-lima/blob/lima-17.3/src/gallium/drivers/lima/lima_screen.c#L325

For mesa there are many:
https://www.mesa3d.org/envvars.html