yuq/mesa-lima #53: Texture alignment and layout problem

yuq, 2018-05-02:
Dump the mali with:
https://github.com/yuq/gfx/tree/master/x11/fbo

Output:
https://gist.github.com/yuq/a17350bd5bf91dbeba24e9deeda9a829
https://gist.github.com/yuq/a32a5ed635bfffe27c49b936a06c4887

Seems the texture memory will be reused and no copy:
1. frame 0 output to 0x10100000
2. frame 1 get texture from 0x10100000
3. 0x10100000 memory not changed from the end of frame 0 to beginning of frame 1

The memory content at 0x10100000 is not linear and the layout field in pp frame wb0 and texture desc are set. Maybe some tiled format or mipmap?

Need more investigation.

anarsoul, 2018-05-03:
Looks like a tiled format.

yuq, 2018-05-13:
Help from Alyssa:
"
Utgard and Midgard have identical formats for tiled textures. So, I
modified the limare code for texture tiling to improve the flexibility
and maybe performance. It might be useful for lima.
https://gitlab.freedesktop.org/panfrost/panloader/blob/develop/trans/limare-swizzle.c
"

anarsoul, 2018-05-13:
We also need untiling routine. Basically we need to plug tiling/untiling routines into lima_transfer_map() and lima_transfer_unmap(). Scanout buffer should be kept untiled, but it cannot be used as texture later (at least not without a copy).

yuq, 2018-05-14:
Yes, we need it anyway. If mali GPU has a format converter DMA, we can use it too. I see in mali kernel driver there's DMA IOREG but never used.

I still doubt that mali doesn't support aligned linear texture. Maybe only because we haven't found where's the setting. I'm thinking if there's a case that mali binary driver has to use an aligned linear texture so that we can write a test for dump.

anarsoul, 2018-05-14:
Tiled textures should be faster, so we'll have to support them anyway. As for aligned linear - limare comment says that proprietary driver uses linear format for large textures. So could you try running your demo with 1300x1300 texture?

yuq, 2018-05-14:
Tried 1300x1300, same tailed texture. I think I have to setup a mali driver supporting gbm interfaces for this.

anarsoul, 2018-05-14:
@yuq could you try setting layout to 1 and 2 in lima_update_tex_desc() in lima? I don't have access to my setup atm to test.

shadeslayer, 2018-05-15:
I tried running weston and Plasma with layout = 1 and 2 ( as well as setting flag0 to both 0 or 1 ).

With Weston, on all 4 configurations, I got nothing on the screen.
With Plasma, on all 4 configurations, I got a garbled cursor on the screen.

So perhaps linear textures don't work?