yuq/mesa-lima #54: License update for mesa-lima and syscall-tracker

yuq, 2018-05-03:
First let me apologize for the previous license violation in the mesa-lima and syscall-tracker code to Luc Verhaegen. As his request, I updated the license in all the place which directly or in-directly copy the lima-ng code by **rebase** in all branches:
mali-syscall-tracker/main.c
mesa-lima/src/gallium/drivers/lima/lima_draw.c
mesa-lima/src/gallium/drivers/lima/lima_texture.c
mesa-lima/src/gallium/drivers/lima/lima_state.c
mesa-lima/src/gallium/drivers/lima/lima_screen.c
mesa-lima/src/gallium/drivers/lima/lima_context.c

Please re-sync the code for the updated license. Sorry for the inconvenience caused by this.

yuq, 2018-05-08:
Add two new files for license update:
mesa-lima/include/drm-uapi/lima_drm.h
linux-lima/include/uapi/lima_drm.h

For data structure from lima-ng:
drm_lima_m400_gp_frame
drm_lima_pp_frame_reg
drm_lima_pp_wb_reg

So did another rebase for all branches of linux-lima and mesa-lima. This also remind me it's bad idea to include register definition in this API file, will remove them in the future, but license update still has to be done for now.

koenkooi, 2018-05-08:
speaking of rebasing, what's your view on rebasing onto the 18.0 branch (to get 18.0.3) or even the master branch?

yuq, 2018-05-08:
Currently I just do rebase for every mesa release (18.0, 18.1). As this driver is still in quite incomplete state, rebase often to get stable fixes (18.0) or new interfaces (master) have less meaning. I'd like to focus on implementing more functionalities. But I think the frequency will increase when the driver gets more completed like the kernel part.

koenkooi, 2018-05-08:
That makes sense, thanks for the explanation!

closed 2018-05-17 by yuq