yuq/mesa-lima #57: Qt5 apps on exynos4412 using eglfs crash

joyqat, 2018-05-28:
this is the console output, can't figure out why. is this because of misconfiguration?

```
Failed to move cursor on screen VGA1: -14
lima_resource_create: pres=0x394a0 width=65536 height=1 depth=1 target=0 bind=10 usage=3
lima_pipe_flush: flags=0
lima_pipe_flush: do nothing
lima_pipe_flush: flags=0
lima_pipe_flush: do nothing
lima_resource_create: pres=0x769e0 width=800 height=480 depth=1 target=2 bind=18000a usage=0
lima_surface_create: pres=0x769e0 psurf=0x76a88
lima_screen_get_compiler_options
lima_resource_create: pres=0x9d848 width=800 height=480 depth=1 target=2 bind=a usage=0
lima_transfer_map: pres=0x9d848
lima_set_framebuffer_state
psurf color=0x76a88 z=(nil)
lima_flush: do nothing
fb dim change tiled=50/30 block=25/15 shift=1/1
lima_clear
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_resource_create: pres=0x323558 width=72 height=1 depth=1 target=0 bind=10 usage=0
lima_transfer_map: pres=0x323558
lima_resource_create: pres=0x326830 width=48 height=1 depth=1 target=0 bind=10 usage=0
lima_transfer_map: pres=0x326830
lima_create_depth_stencil_alpha_state
depth enable=0 min_b=0.000000 max_b=0.000000
lima_bind_depth_stencil_alpha_state
lima_create_fs_state
ppir: unsupported nir_op 98
ppir: unsupported nir_op 14
ppir: unsupported nir_op 14
Segmentation fault
# ./rasterwindow
Failed to move cursor on screen VGA1: -14
lima_resource_create: pres=0x394a0 width=65536 height=1 depth=1 target=0 bind=10 usage=3
lima_pipe_flush: flags=0
lima_pipe_flush: do nothing
lima_pipe_flush: flags=0
lima_pipe_flush: do nothing
lima_resource_create: pres=0x769e0 width=800 height=480 depth=1 target=2 bind=18000a usage=0
lima_surface_create: pres=0x769e0 psurf=0x76a88
lima_screen_get_compiler_options
lima_resource_create: pres=0x9d828 width=800 height=480 depth=1 target=2 bind=a usage=0
lima_transfer_map: pres=0x9d828
lima_set_framebuffer_state
psurf color=0x76a88 z=(nil)
lima_flush: do nothing
fb dim change tiled=50/30 block=25/15 shift=1/1
lima_clear
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_screen_get_compiler_options
lima_resource_create: pres=0x324640 width=72 height=1 depth=1 target=0 bind=10 usage=0
lima_transfer_map: pres=0x324640
lima_resource_create: pres=0x323898 width=48 height=1 depth=1 target=0 bind=10 usage=0
lima_transfer_map: pres=0x323898
lima_create_depth_stencil_alpha_state
depth enable=0 min_b=0.000000 max_b=0.000000
lima_bind_depth_stencil_alpha_state
lima_create_fs_state
ppir: unsupported nir_op 98
ppir: unsupported nir_op 14
ppir: unsupported nir_op 14
Segmentation fault
```

yuq, 2018-05-28:
Seems due to ppir has un-implemented instructions:
ppir: unsupported nir_op 98
ppir: unsupported nir_op 14
which should be:
nir_op_bcsel
nir_op_fnoise3_4

mesa-lima is still in early dev stage, only some simple tests can run with it.

joyqat, 2018-05-28:
thanks for your reply, and sorry for the noise @yuq

closed 2018-05-28 by joyqat

yuq, 2018-05-28:
It's OK. Hope someday we are able to support QT.