yuq/mesa-lima #58: Add support for tiled textures

anarsoul, 2018-05-28:
We need to support tiled textures for performance reasons (they're faster to sample for GPU) and as a workaround for issue #53

anarsoul, 2018-05-28:
Initial support lives in my branch: https://github.com/anarsoul/mesa-lima/tree/lima-18.0-tiled-textures

It currently has few issues:
1) Texture uploaded by memcpy-ing data to pointer returned by gbm_bo_map() is garbled - driver tries to use tiled textures when possible and apparently we don't support creating resources with modifiers yet.
2) Tiling/untiling is not optimized yet.

anarsoul, 2018-05-30:
Rebased onto lima-18.1, not it lives in https://github.com/anarsoul/mesa-lima/tree/lima-18.1-tiled-textures

koenkooi, 2018-05-30:
I've built and installed lima-18.1-tiled-textures on an amlogic s905x board, what should I use to test it?

anarsoul, 2018-05-30:
@koenkooi for example you can sample from 500x500 texture. Try https://github.com/anarsoul/kmscube - it won't work without tiled textures

koenkooi, 2018-05-31:
both 'kmscube-anarsoul' and 'kmscube-anarsoul -M rgba' show a spinning cube on the screen. Picture of the rgba one attached.

```
root@libretech-cc:~# kmscube-anarsoul -M rgba
libEGL debug: found extension `DRI_TexBuffer'
libEGL info: found extension DRI_TexBuffer version 2
libEGL debug: found extension `DRI2_Flush'
libEGL info: found extension DRI2_Flush version 4
libEGL debug: found extension `DRI_IMAGE'
libEGL info: found extension DRI_IMAGE version 17
libEGL debug: found extension `DRI_RENDERER_QUERY'
libEGL debug: found extension `DRI_CONFIG_QUERY'
libEGL debug: found extension `DRI2_Throttle'
libEGL debug: found extension `DRI2_Fence'
libEGL debug: found extension `DRI2_Interop'
libEGL debug: found extension `DRI_NoError'
libEGL debug: found extension `DRI_TexBuffer'
libEGL debug: found extension `DRI2_Flush'
libEGL debug: found extension `DRI_IMAGE'
libEGL info: found extension DRI_IMAGE version 17
libEGL debug: found extension `DRI_RENDERER_QUERY'
libEGL info: found extension DRI_RENDERER_QUERY version 1
libEGL debug: found extension `DRI_CONFIG_QUERY'
libEGL info: found extension DRI_CONFIG_QUERY version 1
libEGL debug: found extension `DRI2_Throttle'
libEGL debug: found extension `DRI2_Fence'
libEGL info: found extension DRI2_Fence version 2
libEGL debug: found extension `DRI2_Interop'
libEGL info: found extension DRI2_Interop version 1
libEGL debug: found extension `DRI_NoError'
libEGL info: found extension DRI_NoError version 1
libEGL debug: did not find optional extension DRI_Robustness version 1
libEGL debug: did not find optional extension DRI_FlushControl version 1
libEGL debug: did not find optional extension DRI2_Blob version 1
libEGL debug: No DRI config supports native format 0x20203852
libEGL debug: No DRI config supports native format 0x38385247
libEGL debug: No DRI config supports native format 0x36314752
libEGL debug: No DRI config supports native format 0x34324258
libEGL debug: No DRI config supports native format 0x34324241
libEGL debug: No DRI config supports native format 0x30335258
libEGL debug: No DRI config supports native format 0x30335241
libEGL debug: the best driver is DRI2
Using display 0xe441c60 with EGL version 1.4
===================================
EGL information:
  version: "1.4 (DRI2)"
  vendor: "Mesa Project"
  client extensions: "EGL_EXT_client_extensions EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses EGL_KHR_debug EGL_EXT_platform_wayland EGL_EXT_platform_x11 EGL_MESA_platform_gbm"
  display extensions: "EGL_ANDROID_native_fence_sync EGL_EXT_buffer_age EGL_EXT_image_dma_buf_import EGL_KHR_cl_event2 EGL_KHR_config_attribs EGL_KHR_create_context EGL_KHR_create_context_no_error EGL_KHR_fence_sync EGL_KHR_get_all_proc_addresses EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image EGL_KHR_gl_texture_cubemap_image EGL_KHR_image EGL_KHR_image_base EGL_KHR_image_pixmap EGL_KHR_no_config_context EGL_KHR_reusable_sync EGL_KHR_surfaceless_context EGL_EXT_pixel_format_float EGL_KHR_wait_sync EGL_MESA_configless_context EGL_MESA_drm_image EGL_MESA_image_dma_buf_export EGL_WL_bind_wayland_display "
===================================
OpenGL ES 2.x information:
  version: "OpenGL ES 2.0 Mesa 18.1.0 (git-9d61364d9d)"
  shading language version: "OpenGL ES GLSL ES 1.0.16"
  vendor: "lima"
  renderer: "Mali450"
  extensions: "GL_APPLE_texture_max_level GL_EXT_blend_minmax GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_discard_framebuffer GL_EXT_draw_buffers GL_EXT_draw_elements_base_vertex GL_EXT_frag_depth GL_EXT_map_buffer_range GL_EXT_multi_draw_arrays GL_EXT_occlusion_query_boolean GL_EXT_read_format_bgra GL_EXT_separate_shader_objects GL_EXT_texture_border_clamp GL_EXT_texture_format_BGRA8888 GL_EXT_unpack_subimage GL_KHR_context_flush_control GL_KHR_debug GL_KHR_no_error GL_NV_draw_buffers GL_NV_fbo_color_attachments GL_NV_read_buffer GL_NV_read_depth GL_NV_read_depth_stencil GL_NV_read_stencil GL_OES_EGL_image GL_OES_EGL_image_external GL_OES_EGL_sync GL_OES_compressed_ETC1_RGB8_texture GL_OES_depth24 GL_OES_depth_texture GL_OES_draw_elements_base_vertex GL_OES_element_index_uint GL_OES_fbo_render_mipmap GL_OES_get_program_binary GL_OES_mapbuffer GL_OES_packed_depth_stencil GL_OES_required_internalformat GL_OES_rgb8_rgba8 GL_OES_stencil8 GL_OES_surfaceless_context GL_OES_texture_3D GL_OES_texture_border_clamp GL_OES_texture_npot GL_OES_vertex_array_object GL_OES_vertex_half_float "
===================================
```
[image]

anarsoul, 2018-05-31:
@koenkooi thanks for testing. It should also work with weston once we fix issue #59