yuq/mesa-lima #59: Lima does clear unconditionally for each frame

anarsoul, 2018-05-31:
Lima driver does clear unconditionally for each frame and it breaks EGL_BUFFER_PRESERVED. As result, weston doesn't work properly - screen has to be repainted completely on each draw.

I believe we need to do clear only when lima_clear() is called.

koenkooi, 2018-05-31:
Is weston supposed to output something with the current state of mesa-lima or with using your texture branch? In both cases I just get a black screen on s905x/mali450. A rk3328/mali450 board arrived this week, but I didn't have the time yet to boot that yet to rule out meson-drm related issues.

anarsoul, 2018-05-31:
Yes, it outputs something before you move cursor. Once you move it you'll see black screen with cursor. If you open a terminal (left click in top left corner) - you'll see a terminal window.

yuq, 2018-06-01:
From the dump, mali blob driver will do clear even user does not call glClear when EGL_BUFFER_DESTOYED. But when EGL_BUFFER_PRESERVED, lima_pp_wb_reg[0] and lima_pp_wb_reg[1] will be set to different target address, lima_pp_wb_reg[1] is the real target.

https://gist.github.com/yuq/0df522caf1514b4758623fac36088533
https://gist.github.com/yuq/7a0404ce79fbafc25431967ecf5d62fb

yuq, 2018-06-02:
I have no idea as it's not dumped out, but should be a buffer used for write. Maybe some aux/tmp/mirror buffer? In fact I don't know why there're three wb and what's each wb used for except the single wb case.

anarsoul, 2018-06-02:
I see. I guess we need better dumping tool - one that can actually parse command stream.